brickbots/PiFinder #6: ENT+A turns screen off

rwillsuk, 2023-03-21:
Trying to bring up the other screens by pressing ENT+A and it will take me back to the preview screen. Press it again and the screen turns off. I have to cycle power to get it back on.

brickbots, 2023-03-21:
This sounds like a bug where the status screen crashes getting the IP address of the unit if the hostname is not exactly `pifinder`.... sorry about that oversight, it got me too when I renamed my unit so I could test two at once.  The beta branch has a fix for this, along with some a left-hand mount config option and related code changes.  These updates will be in the main branch shortly, but for now you can switch to the beta branch to see if this resolves your issue.

SSH into the PiFinder and type
```
cd PiFinder
git pull
git checkout beta
```
This should switch you to the beta branch.

brickbots, 2023-03-24:
I think I've finally tracked down the multiple sources of this status screen crash 🎉   Please update to release 1.2.3 for the latest fixes.

Thanks for reporting this and helping me to track down a couple (hopefully rare) crash conditions.

closed 2023-03-24 by brickbots