brickbots/PiFinder #46: Exception logged and  key not accepted on messier catalog screen

mikewagnercmp, 2023-06-27:
Fired up main, using python 3.10.12. navigated to web page, and clicked the "A" until the catalog screen was reached, then clicked the "C" button" until the messier catalog was selected. Then clicked "0" and then "3" when the web interface crashed and multiple exceptions were logged.

Expected result - Type in M031 , pull up the andromeda object.

 ➜  python cd pythonpython3 -m PiFinder.main -fh -k server --camera debug -x
[06/27/2023, 8:50:24.525 AM] starting main
[06/27/2023, 8:50:24.526 AM] 2023-06-27 08:50:24,525 root: DEBUG using debug camera
[06/27/2023, 8:50:24.616 AM] pygame 2.5.0 (SDL 2.28.0, Python 3.10.12)
[06/27/2023, 8:50:24.616 AM] Hello from the pygame community. https://www.pygame.org/contribute.html
[06/27/2023, 8:50:24.721 AM] Write: Starting....
[06/27/2023, 8:50:27.111 AM] Write:    GPS
[06/27/2023, 8:50:27.740 AM] Write:    Keyboard
[06/27/2023, 8:50:28.168 AM] Write:    Camera
[06/27/2023, 8:50:29.178 AM] Write:    IMU
[06/27/2023, 8:50:29.185 AM] Write:    Solver
[06/27/2023, 8:50:29.202 AM] Write:    Integrator
[06/27/2023, 8:50:29.219 AM] Write:    Server
[06/27/2023, 8:50:29.236 AM] Write:    Event Loop
[06/27/2023, 8:50:29.546 AM] 2023-06-27 08:50:29,545 root: DEBUG loading NGC
[06/27/2023, 8:50:29.604 AM] 2023-06-27 08:50:29,603 root: DEBUG loading IC
[06/27/2023, 8:50:29.624 AM] 2023-06-27 08:50:29,623 root: DEBUG loading M
[06/27/2023, 8:50:29.625 AM] 2023-06-27 08:50:29,625 root: DEBUG loading C
[06/27/2023, 8:50:29.627 AM] 2023-06-27 08:50:29,626 root: DEBUG loading Col
[06/27/2023, 8:50:29.629 AM] 2023-06-27 08:50:29,628 root: DEBUG loading SaA
[06/27/2023, 8:50:29.631 AM] 2023-06-27 08:50:29,629 root: DEBUG loading Ta2
[06/27/2023, 8:50:29.969 AM] 2023-06-27 08:50:29,969:tetra3.Tetra3-INFO: Loading database from: /Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/tetra3/default_database.npz
[06/27/2023, 8:50:55.253 AM] 2023-06-27 08:50:55,253 root: DEBUG Catalog>update_object_info(), self.cat_object=None, self.catalog_index=0, self._catalog_item_index=0
[06/27/2023, 8:50:55.254 AM] 2023-06-27 08:50:55,253 root: DEBUG after key_c, catalog index is 1
[06/27/2023, 8:50:59.630 AM] 2023-06-27 08:50:59,629 root: DEBUG Catalog>update_object_info(), self.cat_object=None, self.catalog_index=1, self._catalog_item_index=0
[06/27/2023, 8:50:59.630 AM] 2023-06-27 08:50:59,629 root: DEBUG after key_c, catalog index is 2
[06/27/2023, 8:51:02.150 AM] 2023-06-27 08:51:02,150 root: DEBUG Calling find by designator with:   M ---
[06/27/2023, 8:51:02.150 AM] 2023-06-27 08:51:02,150 root: DEBUG find by designartor, objectnumber is 0
[06/27/2023, 8:51:02.150 AM] 2023-06-27 08:51:02,150 root: DEBUG Catalog>update_object_info(), self.cat_object=None, self.catalog_index=2, self._catalog_item_index=0
[06/27/2023, 8:51:05.219 AM] 2023-06-27 08:51:05,218 root: DEBUG Calling find by designator with:   M --3
[06/27/2023, 8:51:05.219 AM] 2023-06-27 08:51:05,219 root: DEBUG  searching: c['sequence']=1type(c['sequence'])=<class 'int'>
[06/27/2023, 8:51:05.220 AM] 2023-06-27 08:51:05,219 root: DEBUG  searching: c['sequence']=2type(c['sequence'])=<class 'int'>
[06/27/2023, 8:51:05.220 AM] 2023-06-27 08:51:05,219 root: DEBUG  searching: c['sequence']=3type(c['sequence'])=<class 'int'>
[06/27/2023, 8:51:05.220 AM] 2023-06-27 08:51:05,219 root: DEBUG Found c['sequence']=3, at index 2
[06/27/2023, 8:51:05.220 AM] 2023-06-27 08:51:05,219 root: DEBUG Catalog>update_object_info(), self.cat_object={'catalog': 'M', 'sequence': 3, 'obj_type': 'Gb', 'ra': 205.55, 'dec': 28.383333333333333, 'const': 'CVn', 'l_size': ' ', 'size': 16.2, 'mag': 6.4, 'desc': '!!, glob. cl. , eB, vL, vsmbM, st 11...; = M3     \n'}, self.catalog_index=2, self._catalog_item_index=3
[06/27/2023, 8:51:05.350 AM] Process Process-7:
[06/27/2023, 8:51:05.350 AM] Process Process-4:
[06/27/2023, 8:51:05.350 AM] Process Process-6:
[06/27/2023, 8:51:05.350 AM] Traceback (most recent call last):
[06/27/2023, 8:51:05.350 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/runpy.py", line 196, in _run_module_as_main
[06/27/2023, 8:51:05.351 AM]     return _run_code(code, main_globals, None,
[06/27/2023, 8:51:05.351 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/runpy.py", line 86, in _run_code
[06/27/2023, 8:51:05.351 AM]     exec(code, run_globals)
[06/27/2023, 8:51:05.351 AM]   File "/Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/main.py", line 715, in <module>
[06/27/2023, 8:51:05.351 AM]     main(script_name)
[06/27/2023, 8:51:05.351 AM]   File "/Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/main.py", line 550, in main
[06/27/2023, 8:51:05.351 AM]     update_msg = current_module.update()
[06/27/2023, 8:51:05.351 AM]   File "/Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/ui/catalog.py", line 384, in update
[06/27/2023, 8:51:05.352 AM]     aka.draw((0, posy))
[06/27/2023, 8:51:05.352 AM]   File "/Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/ui/ui_utils.py", line 180, in draw
[06/27/2023, 8:51:05.352 AM] Traceback (most recent call last):
[06/27/2023, 8:51:05.352 AM] Traceback (most recent call last):
[06/27/2023, 8:51:05.352 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
[06/27/2023, 8:51:05.352 AM]     self.run()
[06/27/2023, 8:51:05.352 AM] Traceback (most recent call last):
[06/27/2023, 8:51:05.352 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/process.py", line 108, in run
[06/27/2023, 8:51:05.352 AM]     self._target(*self._args, **self._kwargs)
[06/27/2023, 8:51:05.352 AM]   File "/Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/integrator.py", line 147, in integrator
[06/27/2023, 8:51:05.352 AM]     location = shared_state.location()
[06/27/2023, 8:51:05.352 AM]   File "<string>", line 2, in location
[06/27/2023, 8:51:05.352 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/managers.py", line 818, in _callmethod
[06/27/2023, 8:51:05.352 AM]     kind, result = conn.recv()
[06/27/2023, 8:51:05.352 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/connection.py", line 250, in recv
[06/27/2023, 8:51:05.352 AM]     buf = self._recv_bytes()
[06/27/2023, 8:51:05.352 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/connection.py", line 414, in _recv_bytes
[06/27/2023, 8:51:05.352 AM]     buf = self._recv(4)
[... 142 lines cut ...]
[06/27/2023, 8:51:10.358 AM] Traceback (most recent call last):
[06/27/2023, 8:51:10.358 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
[06/27/2023, 8:51:10.358 AM]     self.run()
[06/27/2023, 8:51:10.358 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/process.py", line 108, in run
[06/27/2023, 8:51:10.358 AM]     self._target(*self._args, **self._kwargs)
[06/27/2023, 8:51:10.358 AM]   File "/Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/imu_fake.py", line 55, in imu_monitor
[06/27/2023, 8:51:10.358 AM]     time.sleep(0.1)
[06/27/2023, 8:51:10.358 AM]     res = self._popen.wait(timeout)
[06/27/2023, 8:51:10.358 AM] KeyboardInterrupt
[06/27/2023, 8:51:10.358 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/popen_fork.py", line 43, in wait
[06/27/2023, 8:51:10.358 AM]     return self.poll(os.WNOHANG if timeout == 0.0 else 0)
[06/27/2023, 8:51:10.358 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/popen_fork.py", line 27, in poll
[06/27/2023, 8:51:10.358 AM]     pid, sts = os.waitpid(self.pid, flag)
[06/27/2023, 8:51:10.358 AM] KeyboardInterrupt: 
[06/27/2023, 8:51:10.358 AM] Traceback (most recent call last):
[06/27/2023, 8:51:10.359 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/process.py", line 314, in _bootstrap
[06/27/2023, 8:51:10.359 AM]     self.run()
[06/27/2023, 8:51:10.359 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/multiprocessing/process.py", line 108, in run
[06/27/2023, 8:51:10.359 AM]     self._target(*self._args, **self._kwargs)
[06/27/2023, 8:51:10.359 AM]   File "/Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/pos_server.py", line 74, in run_server
[06/27/2023, 8:51:10.359 AM]     client_socket, address = server_socket.accept()
[06/27/2023, 8:51:10.359 AM]   File "/Users/mikewagner/.pyenv/versions/3.10.12/lib/python3.10/socket.py", line 293, in accept
[06/27/2023, 8:51:10.359 AM]     fd, addr = self._accept()
[06/27/2023, 8:51:10.359 AM] KeyboardInterrupt
[06/27/2023, 8:51:10.565 AM] %                                                                                ➜  python

mikewagnercmp, 2023-06-27:
Edit to add - based on my snooping with a web proxy, I see the command for key "0" being made with a return of success, but the UI image returned does not up date to reflect it - I was trying to type the leading zero for M031
then when i typed the 3 it caused the crash. the 3 was also acknowledged by the API with a success.

mrosseel, 2023-06-27:
this seems to be the error:
```
[06/27/2023, 8:51:05.353 AM] File "/Users/mikewagner/Projects/PiFinder-main 2/python/PiFinder/ui/ui_utils.py", line 222, in layout
[06/27/2023, 8:51:05.353 AM] f"Layouting {self.text=}, {self.textlen=}, {self.pointer=}, {self.counter=}, {self.scrollspeed=}"
[06/27/2023, 8:51:05.353 AM] AttributeError: 'TextLayouterScroll' object has no attribute 'counter'. Did you mean: 'pointer'?
```

mrosseel, 2023-06-27:
solved in #48

mrosseel, 2023-06-27:
this was some merge error from the PR's to main - if you take a look at the error you can apply the fix locally as well, it's short

mikewagnercmp, 2023-06-27:
That did it thanks!

closed 2023-06-27 by mikewagnercmp