brickbots/PiFinder #56: Feature Request: Swap out IMU BNO055 with BNO085

steamraven, 2023-07-11:
Adafruit is now selling the BNO085 with the exact same footprint.  In fact, it is the same hardware, just a different firmware.  The firmware is supposed to be a whole lot more accurate.

https://www.adafruit.com/product/4754

brickbots, 2023-07-11:
Hi Steamraven!  

Thanks for submitting this issue.  I compared the two units when I was doing the initial R&D for the PiFinder and found it to be... less impressive than the description would make it seem.  It does offer some interesting higher-level classification stuff (like tap/shake/step recognition), but the overall stability of the fused quaternions was similar or worse than the vanilla unit in my, admittedly,  limited testing and the extra high-level classification didn't seem useful for the PiFinder.

It is a couple bucks cheaper, which is certainly nice... but the vanilla unit seemed to be available in larger quantities...so I went with that one.  If someone has time to do more thorough testing to see if it really is more accurate, I'd be willing to support both (have to support legacy hardware), but I'm hesitant to add more lines of code to account for the differences in the two units unless there is a compelling case.

The most interesting potential bit for PiFinder use , the UART UVC, has some specific assumptions about the types of motion likely to be experience as it's fine tuned for robotic navigation so fails to properly read altitude angles changes past a point :-/

steamraven, 2023-07-13:
Thanks for the detailed explanation.  I only really mentioned it, because I have seen a lot of posts complaining about the BNO055 loosing precision fast, but that is probably not a problem for the PiFinder, as the plate-solver can re-acquire position.

I love this project, and thank you for all your hard work on it

closed 2023-07-13 by steamraven