brickbots/PiFinder #68: Better WiFi config via web interface

brickbots, 2023-07-30:
It would be wonderful to have a more approachable way to manage wpa_supplicant.conf settings.  A web interface allowing the removal, addition, and re-ordering of Wifi networks would be fantastic.  This would remove the last 'technical' activity that PiFinder owners might need to perform.

brickbots, 2023-08-06:
This could also include changing the access point name to prevent in-the-field conflicts

mrosseel, 2023-08-07:
> This could also include changing the access point name to prevent in-the-field conflicts

you mean that the AP sent out by the pifinder has a unique number to distinguish it? Ties in with the unique number discussion maybe? Or some hash of the mac address for now.

closed 2023-12-05 by brickbots