brickbots/PiFinder #71: Tetra3 doesn't like trees

mrosseel, 2023-08-09:
Either Tetra3 doesn't like trees or there are not enough stars resolved:

[image]

smroid, 2023-12-17:
I've developed an alternative to Tetra3's star extraction and centroiding step, at https://github.com/smroid/StarGate. When applied to the tree image in this bug, it finds all of the real stars with no false stars in the tree clutter, and the resulting centroid list solves with Tetra3.

I would like to include the test image above in the stargate repo's test_data directory. May I have permission to do so, and to whom should I credit that image?

brickbots, 2023-12-17:
Hi Hi!

I've been watching your project with interest, thanks for testing it against this pathological image.  I'm pretty sure @mrosseel took the original if you are looking for permission/attribution.

I'm not super-familiar with rust, but the gRPC interface looks simple enough.  I might see if I can get up to speed enough to get the StarGate installed and running to give it a few tests.

mrosseel, 2024-01-30:
solved by #131

closed 2024-02-20 by brickbots