brickbots/PiFinder #72: Tetra3 doesn't solve a reasonable image

mrosseel, 2023-08-09:
timeout too strict?

```
Aug 10 01:37:47 piDSC python[3004]: 2023-08-10 01:37:47,819 tetra3.Tetra3: DEBUG Trimmed centroid_input shape to: (30, 2)
Aug 10 01:37:47 piDSC python[3004]: 2023-08-10 01:37:47,921 tetra3.Tetra3: DEBUG Timeout reached after: 0.10127441999975417s.
Aug 10 01:37:47 piDSC python[3004]: 2023-08-10 01:37:47,921 tetra3.Tetra3: DEBUG FAIL: Did not find a match to the stars! It took 102 ms.

```
[image]

mrosseel, 2023-08-09:
after changing the timeout, this is the fail error:

```
Aug 10 01:47:25 piDSC python[3479]: 2023-08-10 01:47:25,758 tetra3.Tetra3: DEBUG Trimmed centroid_input shape to: (30, 2)                      Aug 10 01:47:25 piDSC python[3479]: 2023-08-10 01:47:25,899 tetra3.Tetra3: DEBUG Number of nearby stars: 30, total matched: 4
Aug 10 01:47:25 piDSC python[3479]: 2023-08-10 01:47:25,900 tetra3.Tetra3: DEBUG Mismatch probability = 3.70e-03, at FOV = 10.09174deg
Aug 10 01:47:26 piDSC python[3479]: 2023-08-10 01:47:26,042 tetra3.Tetra3: DEBUG FAIL: Did not find a match to the stars! It took 285 ms.
Aug 10 01:47:26 piDSC python[3465]: 2023-08-10 01:47:26,375 picamera2.picamera2: DEBUG Execute job: <picamera2.job.Job object at 0xe3c10838>
Aug 10 01:47:26 piDSC python[3479]: 2023-08-10 01:47:26,413 tetra3.Tetra3: DEBUG Got solve from image with input: (<PIL.Image.Image image mode=
RGB size=512x512 at 0xE4D02F28>, 10.2, 0.5, 8, 0.01, 0.001, 1000, None, {})
Aug 10 01:47:26 piDSC python[3479]: 2023-08-10 01:47:26,422 tetra3.Tetra3: DEBUG Image (height, width): (512, 512)
Aug 10 01:47:26 piDSC python[3479]: 2023-08-10 01:47:26,502 tetra3.Tetra3: DEBUG Found this many centroids, in time: (66, 78.92169600017951)
Aug 10 01:47:26 piDSC python[3479]: 2023-08-10 01:47:26,503 tetra3.Tetra3: DEBUG Got solve from centroids with input: (66, (512, 512), 10.2, 0.
5, 8, 0.01, 0.001, 1000, None)
Aug 10 01:47:26 piDSC python[3479]: 2023-08-10 01:47:26,503 tetra3.Tetra3: DEBUG Set threshold to: 4.987386898533609e-10, have 2005058 patterns

```

gustavmpettersson, 2023-08-10:
The same database as before, built with `pattern_stars_per_fov=15` does indeed solve the image, however it is 50MB instead of 15...

mrosseel, 2023-08-10:
thanks for the quick response! We'll have to see what the speed impact is of both proposed solutions, but it will in any case be faster than waiting 10 mins for the star pattern to change enough so it resolves ;)

brickbots, 2023-08-12:
Latest release uses new DB with `pattern_stars_per_fov = 15`.  Solves the provided image without difficulty and rather quickly!

closed 2023-08-12 by brickbots