brickbots/PiFinder #75: Using the push nearest function before first solve crashes UI

brickbots, 2023-08-14:
Kiwi72 reported:

[I selected] 5 Push All in Catalog options screen when running on the table inside so obviously no solves and it just blanked the screen or shutdown, hard to tell what happened as I couldn't get the screen back up no matter what buttons or combination I pushed

mrosseel, 2023-08-14:
which version?

brickbots, 2023-08-14:
Kiwi72 said they updated after the 1.6.2 release, so I'm assuming 1.6.... but I'm asking just to double check!

brickbots, 2023-08-14:
Yes, confirmed it was with release 1.6.2!

mrosseel, 2023-08-16:
solved for main branch in #77

mrosseel, 2023-08-16:
should this be backported to release?

brickbots, 2023-08-17:
I think it's sort of edge-case-y, so it can wait until next release?  Let me know if you feel differently and I'll back-port it or do another point release 👍

mrosseel, 2023-08-23:
also backported to release with #78