brickbots/PiFinder #85: Star identification fails with bright stars

mrosseel, 2023-09-08:
On main the camera preview can now detect stars and align on them, but it fails to detect bright stars or, as in this case, Jupiter.
Here  is the solver_debug_dump acquired with ENT+D

[c56a5c9e.zip](https://github.com/brickbots/PiFinder/files/12554459/c56a5c9e.zip)

brickbots, 2023-09-08:
Thank you @mrosseel for the debug info!  I've tried a few different tetra3 star extraction params and I think I've found a set that works well on the example images, is not too slow, and hopefully will be a bit more stable from frame to frame.

Please try the `snap_fixes` branch and let me know what you think 👍

brickbots, 2023-09-19:
I tested this quite a bit over the last week and I think it's solved.  I'm going to update the snap_fixes branch and get a PR going

closed 2023-09-19 by brickbots