brickbots/PiFinder #91: Add Barnard dark nebula catalog

brickbots, 2023-09-19:
http://cdsarc.u-strasbg.fr/viz-bin/Cat?VII/220A

mrosseel, 2023-09-19:
already mentioned in #27 but ok to track it seperately here.
I'm not adding any more catalogs atm because I feel that pretty soon we'll need a better catalog selection mechanism.
Another option is to disable some of the less used ones to keep it manageable, and later allow this to be configured via the menu - this way we can postpone the catalog selection rewrite for some time

brickbots, 2023-09-19:
D'Oh! Sorry, I missed that it was already included in #27 I'll close this one.  Agreed that the catalog volume is getting large.  The simple fix that you suggested, an option to 'exclude' some catalogs from the rotation should be easy to implement with the current menu system and will do the job for a while :+1:

closed 2023-09-19 by brickbots