brickbots/PiFinder #95: Support for Stellarium json observing lists

brickbots, 2023-09-22:
It does not support .skylist export 😞

Baldaccchino, 2023-09-23:
Issue with "updating" directory. Uploading file or making changes on the Samba drive does not immediately reflect when going through the menus to "load." 
e.g. 
NorthSky.skylist filename changed to "NSky.skylist" on Samba.  On the menu it does not reflect the name change.
If I add a file or remove a file changes aren't reflected in menu until after reboot

Also any improperly built lists will cause screen to blank and essentially freezes it; subsequently needing to reboot.

brickbots, 2023-09-30:
> Issue with "updating" directory. Uploading file or making changes on the Samba drive does not immediately reflect when going through the menus to "load." e.g. NorthSky.skylist filename changed to [...]

Both of these issues are fixed in the upcoming release, but it still only supports .skylist