brickbots/PiFinder #100: Allow usernames other than 'pifinder'

hovissimo, 2023-09-29:
The install script hard-codes some paths for the `pifinder` username.

There is a conversation in Discord here: https://discord.com/channels/1087556380724052059/1087556507933089874/1157420010860445737

hovissimo, 2023-09-29:
It seems that I can't add a branch to open a PR, but here's the commit I would submit for this issue.

```
Author: Hovis <github@bunkercode.com>
Date:   Fri Sep 29 14:15:58 2023 -0700

    Use $HOME in pifinder_setup.sh

    This abstracts the username and prevents errors when users pick a
    username when using the Raspberry Pi imager.

    Resolves #100

diff --git a/pifinder_setup.sh b/pifinder_setup.sh
index a693a56..cb14fe2 100644
--- a/pifinder_setup.sh
+++ b/pifinder_setup.sh
@@ -6,27 +6,27 @@ cd PiFinder
 sudo pip install -r requirements.txt

 # data dirs
-mkdir ~/PiFinder_data
-mkdir ~/PiFinder_data/captures
-mkdir ~/PiFinder_data/obslists
-mkdir ~/PiFinder_data/screenshots
-mkdir ~/PiFinder_data/solver_debug_dumps
-mkdir ~/PiFinder_data/logs
-chmod -R 777 ~/PiFinder_data
+mkdir $HOME/PiFinder_data
+mkdir $HOME/PiFinder_data/captures
+mkdir $HOME/PiFinder_data/obslists
+mkdir $HOME/PiFinder_data/screenshots
+mkdir $HOME/PiFinder_data/solver_debug_dumps
+mkdir $HOME/PiFinder_data/logs
+chmod -R 777 $HOME/PiFinder_data

 # Wifi config
-sudo cp ~/PiFinder/pi_config_files/dhcpcd.* /etc
-sudo cp ~/PiFinder/pi_config_files/dhcpcd.conf.sta /etc/dhcpcd.conf
-sudo cp ~/PiFinder/pi_config_files/dnsmasq.conf /etc/dnsmasq.conf
-sudo cp ~/PiFinder/pi_config_files/hostapd.conf /etc/hostapd/hostapd.conf
-echo -n "Cli" > ~/PiFinder/wifi_status.txt
+sudo cp $HOME/PiFinder/pi_config_files/dhcpcd.* /etc
+sudo cp $HOME/PiFinder/pi_config_files/dhcpcd.conf.sta /etc/dhcpcd.conf
+sudo cp $HOME/PiFinder/pi_config_files/dnsmasq.conf /etc/dnsmasq.conf
+sudo cp $HOME/PiFinder/pi_config_files/hostapd.conf /etc/hostapd/hostapd.conf
+echo -n "Cli" > $HOME/PiFinder/wifi_status.txt
 sudo systemctl unmask hostapd

 # Samba config
-sudo cp ~/PiFinder/pi_config_files/smb.conf /etc/samba/smb.conf
+sudo cp $HOME/PiFinder/pi_config_files/smb.conf /etc/samba/smb.conf

 # Hipparcos catalog
-wget -O /home/pifinder/PiFinder/astro_data/hip_main.dat https://cdsarc.cds.unistra.fr/ftp/cats/I/239/hip_main.dat
+wget -O $HOME/PiFinder/astro_data/hip_main.dat https://cdsarc.cds.unistra.fr/ftp/cats/I/239/hip_main.dat

 # Enable interfaces
 echo "dtparam=spi=on" | sudo tee -a /boot/config.txt
@@ -35,7 +35,7 @@ echo "dtparam=i2c_arm_baudrate=10000" | sudo tee -a /boot/config.txt
 echo "dtoverlay=pwm,pin=13,func=4" | sudo tee -a /boot/config.txt

 # Enable service
-sudo cp /home/pifinder/PiFinder/pifinder.service /etc/systemd/system/pifinder.service
+sudo cp $HOME/PiFinder/pifinder.service /etc/systemd/system/pifinder.service
 sudo systemctl daemon-reload
 sudo systemctl enable pifinder```

brickbots, 2024-01-21:
Closing in prep for new deployment/update system in Bookworm release

closed 2024-01-21 by brickbots