brickbots/PiFinder #129: Investigate contrast reserve

mrosseel, 2024-01-25:
In deepskylog.be, they implemented a Contrast Reserve.
This takes into account your local SQM, telescope and magnification to calculate how easily visible an object would be.

Here is the PHP implementation:
https://github.com/DeepskyLog/laravel-astronomy-library/blob/edf28c1ff103dc0822a72b7902ba0c00fba8ba89/src/deepskylog/AstronomyLibrary/Targets/Target.php#L1287

brickbots, 2024-01-26:
This is a really cool concept.  I very much like the idea of a dynamic visibility measure and I've been dreaming of calculating SQM based on the camera capture.  I'm pretty sure this is possible (there are cell phone apps which do this) and should be reliable since the imaging system is very uniform (unlike cell phones).

WimDeMeester, 2024-07-29:
Hi @mrosseel , @brickbots 
I will create a python library to calculate the contrast reserve of a given object (how easy / difficult / impossible it is to see), given the SQM, the telescope diameter and the magnification (or eyepieces).  
Of course, some changes will be needed in the piFinder software too.  I would be good if:
- The SQM is calculated from the solved images (see #133).
- The telescope (focal length) can be added in the piFinder.
- A list of eyepieces can be added in the piFinder.

I would be wonderful if different configurations (telescope / eyepiece combinations) could be saved (and used) in the PiFinder software.
I will probably only have time to work on the python library in September...

mrosseel, 2024-07-30:
hi @WimDeMeester great to hear you're planning to work on the contrast reserve! 
As you say there are some missing pieces that need to come together to make this work. Telescope and eyepiece combos have been on the wish list for a while (to scale the images we show per object for instance), contrast reserve might be the push that was needed.
Regarding timeframe, there are some features we want to finish over the summer so September sounds fine :)

WimDeMeester, 2024-08-01:
Ok, I did find some time this week.
I have created a python package to calculate the contrast reserve for an object.  The package can be found here:

https://github.com/DeepskyLog/pydeepskylog

Installation is possible using 

`pip install pydeepskylog`

Updating PiFinder to use this library will be for September (earliest).  I will need to think a bit on how to integrate this in the best way into PiFinder and I will probably need some help too :P

brickbots, 2024-08-03:
I'm very excited about this concept and implementation @WimDeMeester !  One of the things that has been on the PiFinder todo list for a while is to allow people to configure parameters of their scope and eyepieces.  We were planning on using this information for:
* Making observation logging more comprehensive
* Presenting catalog object images with FOV's that exactly match the eyepiece, rather than the preset 1, 0.5, 0.25, 0.12 fov settings

Adding contrast reserve and eyepiece suggestion per object are even better uses for this data, so I'll aim to move this configuration feature up the priority list to support your work in September!

WimDeMeester, 2024-08-05:
I guess I will also be able to add everything that is needed to include the telescope and eyepieces, but I guess @brickbots and @mrosseel already have some ideas how and where to implement this.  
I don't have experience yet with the code base or with the User Interface of the PiFinder.  My PiFinder should arrive this week, so I hope I will gain some experience in the following weeks under the dark skies of south France.