brickbots/PiFinder #133: Get SQM estimate from camera image

mrosseel, 2024-01-28:
do you have any ideas on estimating SQM from the camera image? Would be great to be able to show this to the user and log it in the observations. 

Found this: [pbase](https://www.pbase.com/samirkharusi/image/37608572&exif=Y)
and also this:
[stargazers lounge page 1](https://stargazerslounge.com/topic/370621-sqm-sky-background-measurement-using-an-image-and-the-astap-program/)
and also this, seems pretty straightforward once an image is platesolved
[stargazers lounge page 2](https://stargazerslounge.com/topic/370621-sqm-sky-background-measurement-using-an-image-and-the-astap-program/page/2/)

brickbots, 2024-02-20:
I very much like the approach of using a plate solved image.  It seems to cut down on the variables quite a bit and does not require a dark frame to be taken :-)

If this is fast enough to do, it can be done for every solve... but I expect we may want to be more judicious... perhaps with a keypress on the camera screen?

mrosseel, 2024-02-20:
Depending on how fast we can get it to be it can be done always, every x minutes or on-demand like when logging an observation.

If I understand page 2 correctly it's just this:

> SQM is defined like this - if you had star of some magnitude and you spread it's light on 1 arc second x 1 arc second - and do this with every arc second squared of the sky - that is your brightness.
> 
> Amount of photons (or flux) coming from patch of the sky must be the same as if coming from a single star of certain magnitude - then we say that sky has brightness of given magnitude per arc second squared.
> 
> Given above - it is easy to calculate magnitude of patch of the sky if you can find ratio of flux between it and some star in the image.
> 
> Say you have 20ADU per arc second squared (again - you average per pixel and use plate scale to get per arc second squared value) and you've identified mag9 star in the image which you measured to have 40000ADU.
> 
> Hypothetical star that corresponds to 1x1" patch of the sky would be 40000/20 = 2000 dimmer than our mag9 star.
> 
> If we calculate magnitude of that ratio it will be mag = -2.5 * log_base_10(1/2000) = ~ 8.25mag
> 
> Now we have mag9 star and difference of additional 8.25 magnitude, this means that our sky is 9 + 8.25 = mag17.25 or SQM 17.25

So we need to know the Pixel sum of a detected star in the field (or many stars), the pixel sum of the entire field and the magnitudes of the detected stars. Probably also some indication of the star annulus if we need to calculate the pixel sum ourselves.