brickbots/PiFinder #137: Get GPS coords from phone browser

mrosseel, 2024-02-13:
The implementation of this is ready, but needs the website to be served over HTTPS.

@HeyApos said:
would it be possible to also take the GPS location of the handy, if using the browser based GUI to transfer to the pifinder? This is not a big deal with google geolocation API (http://dev.w3.org/geo/api/spec-source.html). There are a lot of devices, that can do this (e.g. the OnStep program WebGui, Stellarmate, https://artyom-beilis.github.io/astrohopper.html ,...) I find a hole bunch of ready to use code for this: https://www.google.com/search?q=browser+program+get+gps+from+handy
A "one click button" on the main page for setting time/location would be great. So the GPS dongle would be more or less totally optional. 

local cert using letsencrypt:
https://thriveread.com/ssl-localhost-certificate-for-https-local-domain/

apos, 2024-02-14:
Hi Mike, thanks a lot. This is Axel / HeyApos (Discord) from Germany/Ludwigshafen :-)

apos, 2024-03-17:
The problem which most users will have, that they have to manually deploy the self-signed local domain SSL certificate to their browser to use it (step 9 / 10). This is for 127.0.0.1 (localhost). If the local domain is other than localhost, here pifinder.local it has to be publically avaiable (chapter "Let’s Encrypt SSL Certificate for Local Domain (Not Localhost)"), which is not the case for us, as far as I understand.

jscheidtmann, 2024-08-09:
Reading up on this, I believe there's no easy solution to managing an SSL certificate. The most promising scenario (courtesy ChatGPT) would be this: 

 - create an application, that is locally installed on the  users mobile phone.
 - This app get's the user's permission to access the geolocation info and 
 - once it detects that "pifinder.local" is present, sends a request to the PiFinder's server initializing PiFinder's location
 - e.g. upon activation of the app
 
Big drawback is the effort to set this up for both Android and iOS, though...

jscheidtmann, 2024-08-09:
Assuming 
 - we can get an accurate time fix from a connected phone (didn't look into that), and 
 - that the telescope is level so that the IMU reported angle is the angle between an object and the horizon
one can use [celestial navigation](https://en.wikipedia.org/wiki/Celestial_navigation) to estimate the position.

We would have to check to what precision we a) need the position and b) what precision angle measurements need to have to give a "good enough" position: 

You'd move around the telescope to different parts of the sky, platesolving the sky position and measuring the horizont angle of stars using the IMU and then estimate the position from this data. 

Relevant python packages are [pyalmanac](https://pypi.org/project/pyalmanac/) and [Capella](https://pypi.org/project/Capella/). The latter offers an UI where you can plug in measurements and it calculates your position.