brickbots/PiFinder #145: Support for LX200 align command

brickbots, 2024-02-22:
In our current implementation this function in SkySafari or other LX200 control apps acts like a go-to command and starts a push-to cycle on the PiFinder.

For some use cases where the telescope pointing position is far outside the camera FOV, this could server as an alternate alignment method.  It should cause the PiFinder to store the offset between the provided RA/DEC and current solution and apply this to future solutions.  

Some notes:
* The provided RA/DEC would need to be converted to a sort of camera space offsets as the actual pointing offset will be in the the 2d camera plane (like the current bright star alignment)
* There is likely some point where this offset will be too great to reliably provide telescope pointing position.  Maybe we need a warning for this?

mrosseel, 2024-05-14:
how would you propose to do the push-to if the align command can't be used?
I would prefer to do the align on the pifinder itself.
If both are possible that's of course the best solution.

brickbots, 2024-05-14:
I see this as an alternative to the built in bright star alignment and potential object alignment.  Basically just another way to tell the PiFinder 'Hey the telescope is pointed here!'

mrosseel, 2024-05-14:
That would be fine, my question is how will you replace the push-to feature which is now using the align command?

brickbots, 2024-05-14:
Not sure how it's implemented behind the scenes, but in SkySafari there are two buttons, one for Align and one for GoTo (see image).   We currently do the same thing on both buttons, but I suspect SkySafari is sending different commands for align and GoTo, but I'm not sure.  

The user requested that we use the Align button to do sort of what it does on the LX200, which is to snap the telescope reticle on to the selected item.  Similar to our current Bright Star Alignment.

This might not be a useful enough feature to implement, especially if we implement object alignment ourselves.

[image]