brickbots/PiFinder #177: Add an adapter for Rigel Quikfinder base

jscheidtmann, 2024-05-30:
I am going to add an adapter for a Rigel Quikfinder base:

 * create directories case/adapters/quikfinder
 * change case/README.md
 * reflect this change in documentation under "Mounting".

Shortcoming of this adapter: One cannot orient the PiFinder perpendicular to the ground, as recommended.

Looks like this: 
[image]

brickbots, 2024-05-30:
This is a great idea/addition.  What software are you using to model this in?  Perhaps we can merge the pivot portion of the existing foot with this connector?

jscheidtmann, 2024-05-30:
I used OnShape, free version. I believe you're using Fusion360?

jscheidtmann, 2024-06-01:
Here‘s the first construction:
[image]

Idea is to to be compatible to the dovetail top. Will print it tomorrow, to check dimensions.

PR #178 (2024-06-02): Adapter quikfinder
 * Creates a adapters/quikfinder folder in ./case
 * Quikfinder direct adapter, with optional orientation compatible with dovetail_top
 * Added "Rigel Quikfinder Assembly" section to Build Guide -> Printed Parts -> Mounting

Will need to check links after merge.

Fixes #177
closed 2024-06-02 by brickbots