brickbots/PiFinder #179: More secure WiFi setup

jscheidtmann, 2024-06-02:
As I am going to go to a star party in September, I was wondering how to give PiFinder a better default security posture: 

First, the documentation should recommend to attach an Ethernet cable during the build phase to setup local WiFi client access. This avoids entering WiFi passwords over unencrypted channels, as the documentation currently recommends. 

Second, if the unit activates access point mode, it should:
-  attach a random number (5 decimal places from 00000-99999) to the SSID, such that PiFinders in the vicinity use different SSIDs (at least a collision is rather unlikely), 
- At least WPA2 should be configured on the WiFi, using a random password of 15 or 20 places consisting of characters a-z0-9, in groups of 5. 
- This password should be displayed after boot up and be part of the "main" screens, i.e. it must be accessible from pressing "A" multiple times (screen can be deactivated in settings)
- This configuration shall be created once and be stored and reused by each boot up, 
- until a new configuration is generated by the user in settings.

What do you think?

dnlombard, 2024-06-02:
Yes, /recommend/ Ethernet. There could be much more to it.

Hotspots typically display SSID & PW either on the home screen or from 1 
button press. The former is nicer when the screen will blank after a 
timeout.

WPA2 is fine.

brickbots, 2024-06-02:
Thanks @jscheidtmann for thinking about this and for your passion about security!   I'm on-board with all of your thoughts, except the suggestion of using an ethernet cable.  It seems if your other suggestions were implemented, the need for this would be removed, but I may be missing something.

I have some hints/suggestions for anyone who wants to take this change on, but in the meantime, here are a couple things which can help anyone who is using a PiFinder at a public event:

* The access point name can be changed, and this is probably a good idea to prevent your PiFinder from conflicting with another.   **In a quick PR the documentation can be updated to encourage this** :+1:
*  The user account/Web Interface password can and should be changed.  **Again, a quick PR can adjust the documentation to encourage this.**

These two steps will discourage/prevent any accidental and general mischief.  A determined attacker can still do packet collections and such, but the cost/benefit for this attack would be pretty low as there is not much value to a compromised PiFinder.

**A quick/easy software change could enable manual mitigation of the remaining risks:  Allow the setting of a WPA password on the Access Point network via the web interface.**  Yes, initial configuration is done in the clear, but the chance of an attacker being in the right place at the right time to take advantage of this is, seemingly, quite small.

BUT, better is better and your suggestions are very good ones that should not negatively impact the usability very much.  For anyone who wanted to take a run at implementing some of these, here are some thoughts:

1)  The current PiFinder image is based on an older version of Raspberry Pi OS (Bullseye).  There is a branch where work has been started to port the software to Bookworm. I mention this as the network subsystems have changed significantly between the two so making these changes on Bullseye will require some re-implementation on Bookworm.  Might be worth targeting the Bookworm branch and helping to accelerate the transition there.

2)  There is not any mechanism or hook currently for 'first time boot-up' which could be used for generating the random AP network config.  This would be pretty useful though and there are probably a couple of ways to set this up so that it runs on first boot along with the SD Card volume expansion 🤔

jscheidtmann, 2025-04-04:
On bullseye, having had a look at `hostapd`, this seems quite easy to implement, all one needs to do is add the following lines to hostapd.conf:

```
wpa=2
wpa_key_mgmt=WPA-PSK
wpa_passphrase=<passphrase>
rsn_pairwise=CCMP
```

It turns out, that the more involved seems to be the WiFi country. To make that work, you need to run in addition: 

```
sudo raspi-config nonint do_wifi_country <ISO 2-letter country code>
```

It might then in addition be necessary to also configure some channels and wifi modes, to be recognized by other devices in that country. 

I‘ll work on a pull request next.