brickbots/PiFinder #219: Position & Time for Grab‘n‘Go Use

jscheidtmann, 2024-08-22:
Use Case: An observer at a fixed location wants to use PiFinder for a Grab‘n‘Go telescope.

Requirements:
 1) After switching on, observer position shall be read from the configuration file
 2) Once a GPS lock is achieved, the observer position should be stored in the configuration file (every 10-15 min) and only if the last stored position is different by more then 5 km (?) distance
 3) If PiFinder‘s WiFi is in client mode, time should be synced using ntpd and read at startup from the system.
 4) A configuration option „Grab‘n‘Go“ shall be created, that: 
      4.1 presents position and time at startup and
      4.2 allows to edit position and time
      4.3 upon confirmation the position is stored in the config file

brickbots, 2024-08-23:
Thank you for laying out some actual specifications after the Discord discussion.  This looks good as a general feature set... but I think it using any non-GPS location should always start with some user action.  I.e. we can store and cache the last position, but I'd like choosing to use this to be a step that people actually invoke rather than having it try to do this on start up.

The primary reason being we don't know if the last position is correct and if people don't know the system is using the cached location, they may be confused when things don't work properly.  

Determining the reliability of time is easier as we can see if the unit is online and able to get to an NTP server.. so we can automatically use this data if it's available.

jscheidtmann, 2024-08-24:
Updated with feedback from Brickbots.

jscheidtmann, 2025-06-29:
Shall we create a different issue for "enter position on device" and close this?