brickbots/PiFinder #232: Implement framework to invoke/exercise UI modules in testing environment

brickbots, 2024-09-07:
- Set up a display type that works without hardware or pygame.  I believe the Luma module already has something like this
- Develop a fixture that spins up or mocks all of the items required to invoke a UIModule
  - display_class - the mock/stub display class
  - camera_image - A pil image which can be set to solvable/unsolvable during testing 
  - shared_state - A mock (saved?) shared state where we can set solution/gps info for testing
  - command_queues - This is a dictionary of queue objects, should be easy to mock
  - config_object - Probably just use the actual config object, but could be easily mocked 
  - catalogs - Probably best to use the actual catalogs object
  - add_to_stack - Just a function that returns 
  - remove_from_stack - Just a function that returns
- Add a flatten_menu_items function to menu_manager to traverse the whole tree and return a flat list of menu_item_defininitions
- Develop a UIModule exerciser... 
  - Adjust state in various ways (non solvable image, gps /no gps, ???)
  - Call .activate of module
  - Call all key related methods, with update after each
  - Adjust state and repeat
 
Once we have all of these pieces, then the test becomes
- Set up the fixture
- Get a list of menu_items to test
- Create a class instance for each menu_item
- Pass the class instance to the exerciser

We can add this as the first integration test, but if it's takes too long we can make a new 'slow' category of tests.  In any case, we can then run this on either all merges to main, or perhaps PR's or only PCs

brickbots, 2024-09-07:
https://luma-core.readthedocs.io/en/latest/device.html#luma.core.device.dummy

mrosseel, 2024-09-09:
In general I like this plan, however we should refrain from doing something special for testing, and just make the app more flexible in where the config comes from. Like the queues.
I guess my main point is: let's use this exercise to not only enable testing, but improve the code while we're at it.
One way to check this is if the changes will bring us closer to full replay capability. Not something we need right now, but I feel that would be the correct general direction. 

Also not sure why the stack methods shouldn't do anything?