brickbots/PiFinder #238: Populate_wifi_networks() in sys_utils fails when networks or password contain equal signs

jscheidtmann, 2024-09-15:
Sys_utils.py line 50 uses string.split(„=„) which is wrong and fails, if any line contains additional equal signs.

mrosseel, 2024-09-15:
should probably refactor this to use dbus communication to networkmanager - cleaner than changing config files and we might be able to avoid some reboots

jscheidtmann, 2024-09-16:
Maybe you can create a different ticket for that? 
This one needs fixed, as this kills the process and user(s) on Discord report(s) a hanging PiFinder.

mrosseel, 2024-09-16:
working on it

brickbots, 2024-09-18:
> should probably refactor this to use dbus communication to networkmanager - cleaner than changing config files and we might be able to avoid some reboots

When we finally finish moving to the new version of Raspberry Pi OS, there is a totally new network management system.  I have it roughed in a Bookworm branch, but it's been a long time..... So let's not make any big changes here until we move to the new OS

brickbots, 2024-09-18:
Merged #239

PR #239 (2024-09-19): Fix wifi password with equals sign crash
fixes #238
commit c099cdc8 (2024-09-19): Fix wifi password with equals sign crash (#239)

* Fix #238

* Fix whitespace parsing issues with re

* Remove print statements as per PR review
commit a5e218f0 (2025-01-02): Add option for az arrow reversal (#262)

* Remove align related prints

* GPS async processing (#237)

* Remove active from state, put solver in loop.

Fix typos and nox redefinition error

* Restore logconf

* Crashes on HardwarePWM

* gps rewrite

* wrap the generators

* async gps

* async gps

* merge troubles

* ignore hip_main.dat

* SKY should leave time for TPV to run

* No await for queue ops

* Use other error info for faster lock

* hdop and pdop getting it from sky

* Fix logging

* Fix wifi password with equals sign crash (#239)

* Fix #238

* Fix whitespace parsing issues with re [... cut]
closed 2024-09-18 by brickbots