brickbots/PiFinder #253: Unable to log in using the webinterface when PiFinder is running on a local machine

WimDeMeester, 2024-11-05:
When running the webinterface on my local machine (so not the PiFinder itself), I can not log in anymore...  I get the error

```
Traceback (most recent call last):
  File "/home/wim/micromamba/envs/pifinder/lib/python3.9/site-packages/bottle.py", line 876, in _handle
    return route.call(args)
  File "/home/wim/micromamba/envs/pifinder/lib/python3.9/site-packages/bottle.py", line 1759, in wrapper
    rv = callback(*a, ka)
  File "/mnt/c/Users/u0035065/GitHub/PiFinder/python/PiFinder/server.py", line 149, in login
    if sys_utils.verify_password("pifinder", password):
  File "/mnt/c/Users/u0035065/GitHub/PiFinder/python/PiFinder/sys_utils.py", line 290, in verify_password
    """
NameError: name 'su' is not defined 
```

This can be fixed easily by adding the following 2 lines at line 290 of sys_utils.py:
```
    if not REAL_SYS_UTILS:
        return True

```

jscheidtmann, 2025-01-24:
Hi Wim, I am running into the same problem. 

Well, there's been provisions for that: There once was a `get_sys_utils()`-function that imports sys_utils_fake.py, if `su` cannot be found. Someone seemed to be unware of this and instead catched the exception.

mrosseel, 2025-01-24:
> Hi Wim, I am running into the same problem.
> 
> Well, there's been provisions for that: There once was a `get_sys_utils()`-function that imports sys_utils_fake.py, if `su` cannot be found. Someone seemed to be unware of this and instead catched the exception.

you are correct, the code is in utils.py and should be used to import sys_utils. Since you can reproduce the issue, could you create a PR removing the exception catching?

```
def get_sys_utils():
    try:
        # Attempt to import the real sys_utils
        sys_utils = importlib.import_module("PiFinder.sys_utils")
    except ImportError:
        sys_utils = importlib.import_module("PiFinder.sys_utils_fake")
    return sys_utils
```

jscheidtmann, 2025-01-24:
Can do.

jscheidtmann, 2025-01-26:
Hi Wim, can you check my PR #267 ?

jscheidtmann, 2025-02-13:
Fixed with #267. @brickbots can you close?

brickbots, 2025-02-14:
Absolutely!  Thank you @jscheidtmann :+1:

PR #267 (2025-01-29): remov try/catch around import from sh
fixes #253
closed 2025-02-14 by brickbots