brickbots/PiFinder #257: Docs generation is not automated

jscheidtmann, 2024-12-18:
It seems like a git workflow to build and upload the documentation to readthedocs.io is not in the code repository. 
I looked at the history of the .github folder, but it seems it was never there. 

Do you want me to have a look?

jscheidtmann, 2024-12-18:
Note that the documentation claims that this automation exists.

brickbots, 2024-12-18:
Hi @jscheidtmann.  This is indeed automated, but not in the way you are thinking (build + push).  readthedocs.io has systems building from source in a github.com repo which pull the repo and build via their machinery.  Each time a commit is made to the main or release branch the documentation on readthedocs is updated according to the config in the .readthedocs.yaml file

closed 2024-12-18 by brickbots

jscheidtmann, 2024-12-18:
Argh. Should have checked their docs, first! :-D