brickbots/PiFinder #274: Update should handle downgrade gracefully.

jscheidtmann, 2025-02-08:
When updating, and a newer version is installed, update should at least ask "Are you sure?", before downgrading.

brickbots, 2025-02-08:
Hi @jscheidtmann! When you say 'newer version' are you talking about higher semantic version number in the version.txt file?

Is the use case here for people who are not on the release branch or did you run into a situation where updating from one release to another caused some downgrade?

jscheidtmann, 2025-02-08:
Came across the upgrade dialog in PiFinder while checking for translation. Noticed that there was no indication of a downgrade. 
Being on main/i18n, check was against release branch. 

Given the discussion we had on scripts running at upgrade time, we should make sure downgrading works.

brickbots, 2025-02-08:
Aha, thanks for the clarification.  The current update system is very much one-way-only and there is not a realistic path to making the system as it currently works downgrade capable.

If we want the ability to downgrade, we'll need to switch to a different pattern for managing software releases.  I've designed one which allows safe updating with rollback, arbitrary downgrades, and access to different release branches... but it requires implementation, testing and switching to this new system would require all users to reimage their SD card 😬 

For now I'll make sure the update won't switch you to release if you have manually set yourself to a non release branch :+1:

brickbots, 2025-02-14:
With #276 The software update system will now checks the semver more carefully and will avoid installing previous versions

closed 2025-02-14 by brickbots