brickbots/PiFinder #278: Feature Request: Simulation Mode

jscheidtmann, 2025-02-13:
Let's create a simulation mode: 

-[ ] Flag in settings to enable simulation mode (stored in config)
-[ ] Alt/Az pointing from IMU is converted into RA/DEC at current location and used as platesolve location.

Goal: This way one can experience the user interface (up to Camera and Alignment) by moving the PiFinder in daylight.

brickbots, 2025-02-14:
This seems (mostly?) covered by the existing test mode found under 'Tools'.  Are you thinking of improving this, or completely replacing it?  Two differences I see:

* This proposes that the simulation mode persist across restarts... this has the potential for confusion if a user forgets it's on and starts observing, and only helps if you are restarting multiple times during a simulation session.  Are you thinking this is mostly a tool for developers who may start the software multiple times?

* The current Test system continually resets the position after the PiFinder stops moving.  We could potentially improve this by starting with the image saved from disk and generating a new image synthetically (via the current chart rendering) after the IMU detects motion has started/stopped.  It would make the testing mode more interesting!

jscheidtmann, 2025-02-14:
Good point, I have the feeling test-mode is almost there. I'll look into adjusting that. 

Idea of this is to have a mode, that can be used to demonstrate **all** features of PiFinder.

jscheidtmann, 2025-02-14:
Ok, Test mode seems to do what I want. Sorry to bother you.

closed 2025-02-14 by jscheidtmann