brickbots/PiFinder #286: Logging seems to be split

jscheidtmann, 2025-02-19:
DeadricGems reported a hang after a failed upgrade in Discord at this point:

[image]

But the log file, that we could access via SMB was empty.

I seems that the log configuration is switched / wrong at that point in the boot process.

jscheidtmann, 2025-02-19:
I cannot find that "WiFi:Client" message in the source. Not even find `find . -type f -print0 |xargs -0 grep -i wifi:client`.

brickbots, 2025-02-24:
The splash screen is handled in the file splash.py which is called via a different start up service before networking and other bits are online.

Even if it crashes, the main code should start up, so I don't think having proper logging on the splash screen is needed....

This still does not really help with why the software ended up in an un-usable state during the update and the main.py file crashed before logging anything 🤔   Given that the state of the code/sd card could end up very corrupted if the updated is terminated part way through I'm not surprised it can end up in some broken conditions.

Making the update process more robust, with a fail-safe rollback, is definitely needed

closed 2025-02-24 by brickbots