brickbots/PiFinder #290: Unify Console and Logging

brickbots, 2025-02-24:
This is a proposal to adjust the console such that it reads and displays the log file rather than being a separate source of information.  Originally the console was the only 'log', but since we much better multi-process logging has been implemented, it might make sense to have the console essentially tail the log file.

mrosseel, 2025-02-24:
Maybe rename to Log, and show the logging level as first line of the log. Changing log level there would be nice, maybe too dev centric but could be good for helping to debug issues

jscheidtmann, 2025-03-22:
The primary reason we do logging, is to get info what went wrong, when people report problems. I.e. it’s for customer support. 

What is the primary purpose and customer added value of the console, if we showed the logs there?

I don’t see how it makes sense to show the log on the small screen. I am sorry, but I can imagine people sending us screenshots from the console asking for help. Let’s not do that. 

I think it would make more sense to have a debug or support page in the server, offering to download the logs (i.e. not going through samba).

Also take into account, that when the main ui thread aborts, there is no console: Console is basically a queue, messages of which are captured into a string array in the main ui thread, to be displayed in the console view. The console view in addition is rather dumb at the moment, you can scroll up and down, but not left and right. 

The log contains timestamps, detailed infos and stack traces from all processes, possibly mixed, if two processes happen to log at the same time. Because the log is also basically a queue into which all processes write, but it’s consumed and written to disk in an extra process.

mrosseel, 2025-03-22:
I've been thinking about adding something in the web server under 'Tools' which would zip all the logs and download it. Some warnings about leaking gps coords and ip addresses etc. If someone reports a crash they can do that and send it to us via discord/email. Being able to change the log level would also be nice, put it to debug, recreate the crash and send us the logs.

Now that you mention it, the screen is indeed too small for our kind of logging, maybe it should be removed altogether and replaced by a better status screen. With green/red lights for things such as camera, gps/time, imu. Platesolve could be 'solve within the last minute' for instance.

brickbots, 2025-03-23:
Thanks for the great discussion @jscheidtmann and @mrosseel 🤝  

I think Console probably is not serving much purpose ultimately... and we should either just leave it as is, or remove it.  Adding some logging related tools to the web interface seems like a much better use of someone's time.

I'm going to close this and add an issue with some of the notes here for the web interface

closed 2025-03-23 by brickbots