brickbots/PiFinder #295: Enhancements to make it easier to polar align a platform

alexis-cousein, 2025-03-18:
To polar align a platform (or even a mount!) *roughly* it's easy to use Polaris. (Str 60 in PiFinder parlance). I'll assume the North hemisphere.

-Put the platform at the start of travel (you can put it at the end of travel too, but need to reverse up/down and left_right in the rest if you do)
-Set Polaris as the target.
-Move your scope to Polaris
-Move the platform to the other end of travel.

You've just made the scope point to a new location rotated 15° counterclockwise around the polar axis of the platform. If that was pointing to Polaris, then Polaris will not have moved.

If Polaris is now "up" that means the platform axis is left of Polaris, so it needs to be rotated clockwise
If Polaris is now "down" that means the platform axis is right of Polaris, so it needs to be rotated counterclockwise
If Polaris is now "left" that means the platform axis is under Polaris, so the N side needs to be raised.
If Polaris is now "right" that means the platform axis is above Polaris, so the N side needs to be lowered (or the S side raised).

The good thing is that the target arrows of the PiFinder will actually tell you *exactly* what needs to be done.

This allows you to use Polaris, but the North Celestial Pole is actually 0.7° off.

Alas, the NCP is not one of the possible targets so you can't use the same trick to align more properly on the true pole.

Some possible fixes:
-allow input of target using RA/DEC. The NCP is at Dec +90° at any RA. Might be cumbersome
-A "Celestial Pole" category of objects with just two objects (the NCP and SCP)
-A hack, but one that will work: add numbers 999 (NCP) and 998 (SCP) to the "Star" catalogue

Ideally of course a DIRECT alignment help would be there.

-Point PiFinder anywhere (sure beats having to use a pole)
-Go to "polar alignment help" -- PiFinder plate solves where it is now
-Input  pole (NCP, SCP), angle of rotation and direction (clockwise, counterclockwise) of rotation around mount axis
-Move mount along RA by specified angle (usu whatever the platform will support or for regular mounts 90° or 180°)
-Hit enter.

Remember the parameters and you won't even have to correct for elapsed time most of the time.

The new plate solve's location and old plate solve location can only correspond to the angle the user indicated for the mount rotation for *one* location of the centre of rotation. Display how much in Alt and Az the true pole is from there (with arrows).

alexis-cousein, 2025-03-18:
Actually, if you also know the image rotation you don't even need ANY input. If the polar alignment is off you'd have field rotation between the two views that you can try to detect.

 For mounts that you can rotate over a lot that's good enough, but platforms can only be rotated 10° or so.

brickbots, 2025-03-23:
If we got the N/S CP added as objects, is there anything else that needs to be done here, or is that enough to enable a knowledgable user to polar align their platform?

alexis-cousein, 2025-03-24:
It's not ideal but it's a lot better than it is now.

If you know the platform rotation angle between the two extremes, and you point the scope to the NCP, then you know that the rotation axis is roughly 90° from where the NCP moves to after the platform reset; for the usual angle of 15° you also know that the distance to that rotation axis is roughly four times the distance that the NCP walked over ( from the geometry of two mirrored triangels, tan(7.5°) *1/2). 

So it lets you read the Alt and multiply by 4 to get the Az error and lets you read the Az error and multiply by four to get the Alt error. 

But then you need to play a little visualisation mind game to know in what direction. If your platform did a reset clockwise facing N then you expect the NCP to do a movement counterclockwise around the platform's axis with respect where the scope is now pointing, so if the NCP is now left and up from where the platfom is pointing, then the rotational axis is too high (left->high) and too much to the left (up->left) of the real NCP.

If you could input the direction of rotation of the platform and the angle and let the PiFinder do a before and an after plate solve, you could *directly* determine the axis of rotation (by the location of the two plate solves) and say how much to raise/lower the platform in Alt and how much to rotate it in Az to get alignment correct, without any iteration or any possiblity of guessing the direction wrong. 

And you wouldn't even need to point your scope to the NCP to do it, *ANY* plate solve would allow you to find it. In fact, plate solves far from the NCP would be more precise.

alexis-cousein, 2025-03-25:
Edited last comment for clarity.

ilz-cyber, 2026-02-18:
In the Cedar-server documentation, there is a mention of Polar Alignment (aka polar alignment advice) feature, that might satisfy the feature request. Or at least an initial implementation. However, I’m not sure how accessible this code is to PiFinder UI and I’ve never used in Cedar, so can't verify it's usable code. It appears to be declaration drift based method based on the description.
https://github.com/smroid/cedar-server/blob/main/about.md
https://github.com/smroid/cedar-server/blob/main/polar_align.md

The referenced background material brings up some advanced issues that don't directly impact usability for a visual observer..
https://canburytech.net/DriftAlign/index.html
Those guidelines that suggest the Azimuth correction advice implemented in cedar-server is probably good; but also suggests the Altitude correction probably is only good for visual (not photography) since a declination drift method is prone to atmospheric refraction. It does touch on a polar drift method similar to what has been discussed in March comments.
https://web.archive.org/web/20200114025051/http://www.blandinequatorialplatforms.com/Polar%20Alignment.pdf